DomoticX/domoticz-modbus #3: Eastron Energy meter with modbus tcp-rtu converter

miroslavpetrov, 2018-08-28:
Hello,

I am trying to connect an Eastron SDM230 Meter to Domoticz using KMtronic modbus tcp/rtu bridge. It is working but how can I select the type of virtual device that is added to Domoticz? For example I would like the KWH counter to show as counter so I can track my daily usage. Now all parameters are displayed as custom sensors.

DomoticX, 2018-08-28:
Yup, because this plugin is universal modbus plugin to read/write specific registers, it's not made for a specific device, you probably need a SDM230 plugin ;-)

bonebuster, 2018-08-28:
Hi 

that work fine with SDM120 for me
and HF2211 Modbus TCP/RTU bridge
even if RTU in SDM120 four register for voltage is at 30001 parameter
in TCP mode with HF2211 you have to read input register Function 4, register 0, Data type Float 32-Bit
For current in parameter 30007 in SDM120 with RTU, in TCP Mode with HF2211 you have to read register 6
and it's the same for others data (30013 for active power in RTU, register 12 with HF2211 bridge)
for example

carlb666, 2019-08-08:
DomoticX - Is there such a plugin?

carlb666, 2019-08-08:
See: https://www.domoticz.com/wiki/Eastron_SDM120C
and: https://github.com/gianfrdp/SDM120C

closed 2020-03-27 by miroslavpetrov