DomoticX/domoticz-modbus #4: Modbus sensors showing up as Light switch

thefox154, 2018-10-16:
When i try to add sensors (hitachi aws-02-mbus & Eastron DSM530mbus)  they show up as a light switch in Domoticz. 

[image]

DomoticX, 2018-10-16:
Hi thefox154, yes it seems so...some people have giving me tips to update the code, you can revert it by changing switch to custom in this file: https://github.com/DomoticX/domoticz-modbus/commit/127dcbe6d1af3ae2ffe773b4bf62c74c1601ce46#diff-83c76d80ded8d8ab85ec8daeee8d35a1
I am not using domoticz atm so unfortunatly i cannot troubleshoot user input.

thefox154, 2018-10-19:
Hi sebastiaan,

Super bedankt!!!
Alles werkt nu naar behoren.

Groeten,

Pieter

Sandolution, 2018-11-02:
Should be solved with pull request #5

DomoticX, 2020-05-05:
Fixed in update Modbus READ 2020.2A, you can select sensor type...

closed 2020-05-05 by DomoticX