DomoticX/domoticz-modbus #7: TCP Slave ID

Skulldron, 2019-01-02:
Hello,
I am wondering if it's possible to choose slave ID in TCPIP method if there are few more modbus slaves with different IDs on one IP adress?
Thanks in advance.

PhoenixInteractiveNL, 2019-01-02:
Hello Skulldron,

At the moment you can add one IP and ID per "device", i think if you want more you have to add more devices...

Skulldron, 2019-01-08:
Okay, so here is the screen of config panel:
https://drive.google.com/open?id=1d1RqeC8Tt_gyo41RTjwsUlQ0FiTanW9P

I see device adress entry, where should i enter slave ID?

PhoenixInteractiveNL, 2019-01-13:
Hello Skulldron,

I now see what you are pointing at, it is not possible at the moment, but i will write some code in the next days to implement it....

unto1, 2019-02-02:
for modbus Write, it work for me. thanks for this code.

sebekhtc, 2019-02-04:
I'm trying to get my communication working with Slave ID = 7 on synology dsm but I get:
 2019-02-04 14:35:56.844 Error: (Modbus) failed to load 'plugin.py', Python Path used was '/usr/local/domoticz/var/plugins/modbus-read/:/volume1/@appstore/py3k/usr/local/lib/python35.zip:/volume1/@appstore/py3k/usr/local/lib/python3.5/:/volume1/@appstore/py3k/usr/local/lib/python3.5/plat-linux:/volume1/@appstore/py3k/usr/local/lib/python3.5/lib-dynload'.
2019-02-04 14:35:56.844 Error: (temperature1) Module Import failed, exception: 'TabError'
2019-02-04 14:35:56.844 Error: (temperature1) Import detail: File: /usr/local/domoticz/var/plugins/modbus-read/plugin.py, Line: 248, offset: 64
2019-02-04 14:35:56.844 Error: (temperature1) Error Line ' AddressData = Parameters["Address"].split("/") # Split on "/"
2019-02-04 14:35:56.844 ' 

The previous version had no issue with the python, I could create a device but got no comm as my device is slave id no 7 and  I just got a message
"Modbus error communicating! (RTU/ASCII/RTU over TCP), check your settings!" 

Please advise.

PhoenixInteractiveNL, 2019-02-15:
Maybe is the string type in the command,, for the read section, can you try changing it to int and try again?:

client = ModbusClient(host=UnitAddress, port=int(Parameters["Port"]), unit_id=UnitIdForIp, auto_open=True, auto_close=True, timeout=5)

into:

client = ModbusClient(host=UnitAddress, port=int(Parameters["Port"]), unit_id=int(UnitIdForIp), auto_open=True, auto_close=True, timeout=5)

Note: unit_id=UnitIdForIp >> unit_id=int(UnitIdForIp)

sebekhtc, 2019-02-16:
Update: 
Working at site with Modbus communciating devices I took opportunity to troubleshoot the script.
The read from the modbus slave works with the version with corrected intendation (the one I left in pastebin). No need to correct the value reading part. But I get decoding error.
My other problem is having 12 modbus slaves I can only read 3 of them, two of them are Slave ID =1, one is SlaveID=2, that means the UnitID feature works.
Why I can't reach other slaves while my control system can and I can ping them from my Ubuntu with Domoticz  - I have no idea.  (I use the python scripts straight from the linux terminal, troubleshooting ongoing, if I find something I will update the thread, also waiting for suggestion - pure python decoding procedure that I can tes from the terminal etc.)

sebekhtc, 2019-03-13:
I did another test this time on Modbus RTU over TCPIP and found to get a reading:
1) require to import:
from pymodbus.transaction import ModbusRtuFramer

2) modify at # pymodbus: RTU over TCP changing the below from:
c = ModbusTcpClient(host=UnitAddress, port=int(Parameters["Port"]), timeout=5)

to:
client = ModbusTcpClient(host=UnitAddress, port=int(Parameters["Port"]), timeout=5, framer=ModbusRtuFramer)

3) modify at  "# pymodbus section" changing the below (4 times for each type of read) from:
if (Parameters["Username"] == "1"): result = c.read_coils(int(Parameters["Password"]), registercount, unit=int(UnitAddress))

to (slave ID must be used not the IP address of the device):

if (Parameters["Username"] == "1"): data = client.read_coils(int(Parameters["Password"]), registercount, unit=int(UnitIdForIp))

Decoder still gives exceptions, not doing that while running similar script straight from terminal.

sebekhtc, 2019-03-17:
OK found a reason for the decoder, I had to manually install the components, on DSM synology I had to use:
sudo pip3 install -U pymodbus constants
sudo pip3 install -U pymodbus payload

I also had to install the "serial" that was missing in my Synology DSM unlike to my PC that I use for Domoticz tests:
sudo pip3 install -U pymodbus serial

Domot, 2019-03-20:
Hello,

When I'm polling my Siemens Logo! modbus TCP port with an universal poller it presents data of two analog input values in the input registers 30001 and 30002 (Function 04).

I have downloaded the plugins to show the data in Domoticz. When polling using the Modbs Domoticz plug-in V1.1.3 it gives an error.

I have followed your examples but the logfile in Domoticz shows: Modbus error decoding or recieved no data (TCP/IP)!, check your settings!

Can you give me any advice in what is going wrong?

Below a chronological graphical overview:
[image]

[image]

[image]

[image]

[image]

Domot, 2019-03-22:
My issue has solved by solution of unto1 (commented on 9 Feb) in Issue modbus TCP cant read value.

Thanks a lot for finding out!

jesperrooth, 2019-04-25:
> Hello Skulldron,
> 
> I have written the code, i cannot test it ATM, can you try it out?
> Enter the device ID behind the IP address with a slash, like: 192.168.2.1/56
> 
> DL @ pastebin:
> READ: https://pastebin.com/nUMnZefu
> WRITE: https://pastebin.com/cB9qNM85

Hi

Even though I used your public files from https://github.com/DomoticX/domoticz-modbus/ or the files from pastebin above I get error messages from the new device ID / IP.

Error Line ' AddressData = Parameters["Address"].split("/") # Split on "/"

DomoticX, 2020-05-05:
In update Modbus READ 2020.2A, options IP:PORT have been merged, also option readout is much better now. closing this one...

closed 2020-05-05 by DomoticX