DomoticX/domoticz-modbus #12: Corrected script for proper Modbus RTU over TCP/IP and Modbus TCP/IP

sebekhtc, 2019-04-19:
I corrected the script, now Modbus RTU over TCP and Modbus TCP working together with payload decoder.

(I had to add more imports, plus for the Modbus TCP the value was never updated unless it was a coil)

My version:
https://pastebin.com/eQNd7eGm

sebekhtc, 2019-04-20:
small update:
line 393:            
          Domoticz.Debug("MODBUS DEBUG RESPONSE: " + str(data[0]))
give error: ("Modbus error communicating! (TCP/IP), check your settings!"), but still read the value

replaced with:
         Domoticz.Debug("MODBUS DEBUG RESPONSE: " + str(data))

https://pastebin.com/DUa8x0qC

sebekhtc, 2019-04-20:
Waiting for the author to accept

DomoticX, 2019-04-23:
Added/fixed in v1.1.9

closed 2019-04-23 by DomoticX