DomoticX/domoticz-modbus #13: (Modbus) failed to load plugin.py on  Python3.5.3

cezarysz, 2019-05-06:
Hi
After plugin upgrade from 1.1.3 I have erron in Domoticz:
Error: (Modbus) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/modbus-read/:/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-arm-linux-gnueabihf:/usr/lib/python3.5/lib-dynload:/usr/local/lib/python3.5/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.5/dist-packages'.

Error: ('DEVICE NAME') Module Import failed, exception: 'TabError'
Error: ('DEVICE NAME') Import detail: File: /home/pi/domoticz/plugins/modbus-read/plugin.py, Line: 262, offset: 66
Error: ('DEVICE NAME') Error Line '	AddressData = Parameters["Address"] #.split("/") # Split on "/"

I find that the source of the problem can be in Python version (3.5.3).  I tried to remove Python 3.5 and install only 3.4 version but could not check if it solved problem because after install dependencies it updates to 3.5.

DomoticX, 2019-05-06:
Hello, should be fixed in v1.2.0, can you try it out?

cezarysz, 2019-05-06:
Now is OK. Thanks
Still don't know why after more than one month of using plugin now I've got :
"Modbus error decoding or received no data (RTU/ASCII/RTU over TCP)!, check your settings!"
Tryed two different Rpi3 B+, and two different modbus converters. I'm workig with ventilation system, modbus interface is fine.

cezarysz, 2019-05-06:
"Tab error" also in modbus-write plugin.

Domot, 2019-05-06:
Modbus write V1.1.1 does also not work with me.

2019-05-06 23:31:17.939 Error: (ModbusDEV-WRITE) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/modbus-write/:/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-arm-linux-gnueabihf:/usr/lib/python3.5/lib-dynload:/usr/local/lib/python3.5/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.5/dist-packages'.
2019-05-06 23:31:17.939 Error: (Modbus TCP write) Module Import failed, exception: 'TabError'
2019-05-06 23:31:17.939 Error: (Modbus TCP write) Import detail: File: /home/pi/domoticz/plugins/modbus-write/plugin.py, Line: 154, offset: 64
2019-05-06 23:31:17.939 Error: (Modbus TCP write) Error Line '	AddressData = Parameters["Address"].split("/") # Split on "/"

What values do I have to fill in for Payload on & - off when I want to write to a single coil?

cezarysz, 2019-05-07:
> What values do I have to fill in for Payload on & - off when I want to write to a single coil? " <
For me it's HEX value (1,2,3,4,5,6,7,8,9,A,B.....)

DomoticX, 2019-05-07:
> > What values do I have to fill in for Payload on & - off when I want to write to a single coil? "
> 
> For me it's HEX value (1,2,3,4,5,6,7,8,9,A,B.....)

0x01, 0x02 ... 0x0A

DomoticX, 2019-05-07:
> "Tab error" also in modbus-write plugin.

Fixed in v1.1.2, please try it out!

cezarysz, 2019-05-07:
Still problem in modbus-write:

> 2019-05-07 21:50:05.093 Error: (ModbusDEV-WRITE) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/modbus-write/:/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-arm-linux-gnueabihf:/usr/lib/python3.5/lib-dynload:/usr/local/lib/python3.5/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.5/dist-packages'.
> 2019-05-07 21:50:05.094 Error: (xxx) Module Import failed, exception: 'TabError'
> 2019-05-07 21:50:05.094 Error: (xxxa) Import detail: File: /home/pi/domoticz/plugins/modbus-write/plugin.py, Line: 180, offset: 111
> 2019-05-07 21:50:05.094 Error: (xxx) Error Line '	client = ModbusTcpClient(host=UnitAddress, port=int(Parameters["Port"]), timeout=5, framer=ModbusRtuFramer)
> 2019-05-07 21:50:05.094

DomoticX, 2019-05-07:
Fixed try again...

Domot, 2019-05-07:
[image]
[image]

I used 0x00 & 0x01 for payload off & on, the previous errors have now changed to one error. 

At the very first time I configured the hardware for modbus-write in Domoticz the switch  in devices. But after deleting it in Devices it never came back after removing and adding new hardware. Any idea?

cezarysz, 2019-05-07:
> Fixed try again...

Now it's OK. Thanks.

cezarysz, 2019-05-07:
> 2019-05-07 21:59:36.793 Status: (xxx) Initialized version 1.1.2, author 'S. Ebeltjes / domoticx.nl'
> 2019-05-07 21:59:36.792 Status: (xxx) Entering work loop.
> 2019-05-07 21:59:46.796 (xxx) Pushing 'onHeartbeatCallback' on to queue
> 2019-05-07 21:59:46.814 (xxx) Processing 'onHeartbeatCallback' message
> 2019-05-07 21:59:46.814 (xxx) Calling message handler 'onHeartbeat'.
> 2019-05-07 21:59:56.798 (xxx) Pushing 'onHeartbeatCallback' on to queue
> 2019-05-07 21:59:56.836 (xxx) Processing 'onHeartbeatCallback' message
> 2019-05-07 21:59:56.837 (xxx) Calling message handler 'onHeartbeat'

Any idea if I'm in that kind of loop in modbus-write??

DomoticX, 2019-05-07:
No, maybe debug is on?, maybe DomoticZ version not yet supported? hopefully someone can help, i don't use domoticz anymore (instead using node-red)

closed 2019-05-10 by cezarysz

Domot, 2019-05-11:
Tested V1.1.2 for function  5 (write single coil) with payload.  ON = 1 and payload OFF = 0 it works fine.