DomoticX/domoticz-modbus #17: Read Relay Status - RTU

grasuoare, 2019-07-22:
Hi there. I'm trying to read the status of the relays trough RTU Mode and i get the following error logs:

 2019-07-22 21:49:47.820 Debug: SQLH UpdateValueInt Internal Temperature HwID:6 DevID:1 Type:80 sType:5 nValue:0 sValue:37.6
2019-07-22 21:49:47.841 Debug: SQLH UpdateValueInt CPU_Usage HwID:6 DevID:0000044D Type:243 sType:6 nValue:0 sValue:3.48 

Could you help me with this issue?
[image]

cezarysz, 2019-07-23:
Did you try to change sensor type to "custom"? What's the value of sensor then?

grasuoare, 2019-07-23:
with custom is the same: 
 2019-07-23 14:43:12.156 (test) Modbus error decoding or received no data (RTU/ASCII/RTU over TCP)!, check your settings!
2019-07-23 14:43:12.157 Debug: SQLH UpdateValueInt test - HwID:7 DevID:00070001 Type:243 sType:31 nValue:0 sValue:0

cezarysz, 2019-07-23:
Sorry I've similar problem with devices. I sure this is script problem.

cezarysz, 2019-07-23:
I solved my problem by adding data.registers in pymodbus section
try it in decoder you use
https://github.com/DomoticX/domoticz-modbus/pull/16

wt., 23 lip 2019 o 13:54 Cezary Szymanski <cezarysz84@gmail.com> napisał(a):

grasuoare, 2019-07-23:
i tried adding .registers for the 3 lines mentioned and now i get:

2019-07-23 20:23:18.039 (test) Modbus error decoding or received no data (RTU/ASCII/RTU over TCP)!, check your settings!
2019-07-23 20:23:18.040 Debug: SQLH UpdateValueInt test - HwID:7 DevID:00070001 Type:243 sType:31 nValue:0 sValue:0

grasuoare, 2019-07-23:
p.s at every reading i see that the communication led from the relay board blinks.

grasuoare, 2019-07-24:
Guys, any news about this? could somebody tell me what to do or what am i doing wrong? Thanks in advance

cezarysz, 2019-07-24:
Try to change data type. I think the problem is in data convertion for
"noco".

cezarysz, 2019-07-24:
In my case  Modbus RTU/ASCII/TCP - READ v1.1.3 is working good when reading
"No conversion (1register) data type". Please try it,

śr., 24 lip 2019 o 21:42 Cezary Szymanski <cezarysz84@gmail.com> napisał(a):

grasuoare, 2019-07-25:
i tried all data types. same result.

cezarysz, 2019-07-26:
> i tried all data types. same result.

Did you tryed to read "no conversion data type register" using plugin v1.1.3?

grasuoare, 2019-07-29:
Hi cezarysz,

I just added the plugin v1.1.3. Same error.  Check image with the settings:
[img]https://user-images.githubusercontent.com/20748661/62062410-a3a2ae00-b231-11e9-9f90-5193caad6d33.jpg[/img]

2019-07-29 18:46:37.328 (test4) Modbus error decoding or recieved no data (RTU/ASCII/RTU over TCP)!, check your settings!

Antikillerbm, 2019-07-30:
I got the same error in modbus RTU mode only. 
Even discrete inputs not working

cezarysz, 2019-07-30:
Did you try to read registers using plugin v1.1.3?

Antikillerbm, 2019-07-30:
> […](#)

Not yet. I have no possibility to check now sorry

grasuoare, 2019-07-31:
> 
> 
> Not yet. I have no possibility to check now sorry

We are all waiting. Thank you for this plugin!

DomoticX, 2020-05-05:
In update Modbus READ 2020.2A lot has been changed also added BOOL option, closing this one...

closed 2020-05-05 by DomoticX

awalsum, 2022-10-28:
I have been trying to use the BOOL option, but it isn't working for me. I keep getting errors saying that no data received etc. I am using this plugin to read various INT and FLOAT values with function 1 and 3 and that works well. I also use the Write plugin and that works well too. I have tried to find the cause of the BOOL not working in this script, but can't lay my finger on it.