DomoticX/domoticz-modbus #18: Modbus error decoding or received no data (TCP/IP)!, check your settings! SMA

mike05-1976, 2019-10-19:
I have a problem with communication.

[image]
[image]
wat is  it wat i do wrong

DomoticX, 2019-11-03:
Hi There, I think you should use functions 3 “Read holding registers”.

 

Met vriendelijke groet,

 

Sebastiaan Ebeltjes

 

 <https://user-images.githubusercontent.com/56761476/67145196-a4227200-f27f-11e9-80b5-8fe798d44a4e.png> 
 <https://user-images.githubusercontent.com/56761476/67145308-bea91b00-f280-11e9-932e-9a839d2477e4.png>

DomoticX, 2020-05-05:
Ok, we have installed domoticz 2020.2 and got modbus working again, we are updating the plugin asap, stay tuned!

DomoticX, 2020-05-05:
Fixed in update Modbus READ 2020.2A, if not, please make another topic. closing this one...

closed 2020-05-05 by DomoticX