DomoticX/domoticz-modbus #19: Modbus error decoding or received no data (TCP/IP)!, check your settings!

Baeda73, 2019-12-24:
Hi,

I tried to read data from a Modbus Device via TCP / IP, but get the following messages after turning on Debug information.
2019-12-24 16:19:44.004  (ETA Puffer Oben) MODBUS DEBUG RESPONSE: [0, 664]
2019-12-24 16:19:44.004  (ETA Puffer Oben) Modbus error decoding or received no data (TCP/IP)!, check your settings!

Although I get a correct reading from my device (0, 664) the  conversion fails.
The code fails at:
decoder = BinaryPayloadDecoder.fromRegisters(data.registers, byteorder=Endian.Big, wordorder=Endian.Big)

Any idea to track this further ?

Best regards,

Peter

DomoticX, 2020-05-05:
Ok, we have installed domoticz 2020.2 and got modbus working again, we are updating the plugin asap, stay tuned!

DomoticX, 2020-05-05:
Fixed in update Modbus READ 2020.2A, if not, please make another topic. closing this one...

closed 2020-05-05 by DomoticX