DomoticX/domoticz-modbus #22: ModbusRTU slave address issue

Darek77, 2020-02-28:
Hi,
I'm using 2 slave devices connected to the same ModbusRTU net. 
First device with address 1 second device with address 70. The first one works fine, Domoticz read registers correctly, but when I want to read registers from second one, domoticz read registers from first one. 
Seems like plug in always read from device address 1 (no matter what RTU address I put there)
I'am using release 2.0.0

[image]

DomoticX, 2020-02-28:
Try 2/0

Darek77, 2020-03-01:
70/0 works ok.
thx
I have chacked this with Modbus RTU/ASCII/TCP - WRITE.

DomoticX, 2020-05-05:
In update Modbus READ 2020.2A, the fields IP:PORT are merged and options working much better now!, closing this one...

closed 2020-05-05 by DomoticX