DomoticX/domoticz-modbus #24: Plugin, won't start

Allras82, 2020-03-29:
Hi, 

I'm not able to start the plugin, Domoticz log is saying:
2020-03-29 15:03:06.532 Error: PluginList: Parsing '/home/pi/domoticz/plugins/modbus-read/', 'Error document empty.' at line 0 column 0 in XML ''.
2020-03-29 15:03:06.532 Error: PluginList: Parsing '/home/pi/domoticz/plugins/modbus-write/', 'Error document empty.' at line 0 column 0 in XML ''

Domoticz Version: 2020.1
Python Version: 3.7.3

Can you help?

BR
Allan

DomoticX, 2020-03-30:
Hi Allan,
Actually it's not a bug...as stated: Succesfully Tested on Domoticz version: 3.8153 (Stable)

Your Domoticz version is not supported yet...

So hopefully the community (or we do it later this year) can update the plugin (we're on Node-Red atm)

KoosBoersma, 2020-04-23:
Hello,

Is there any development on new versions??
Also running 2020.01 and Python 3.7.3

Regards,
Koos.

DomoticX, 2020-04-23:
Hi Koos,

As stated above your comment:
There are currently no push-pull actions on the code, we....ourselves have moved to Node-red.

We are developing a new domotica system wich will be released somewhere Q4 2020 and we're planning to add support for Node-red and Domoticz later-on, so the plugin might be updated to the new version ;-)

JeroenEeuwes, 2020-04-27:
Perhaps you can mention on the main page this plugin doesn't work with current versions of domoticz? I guess that'll save some people the time to find out why the plugin doesn't work.

KoosBoersma, 2020-05-04:
I found some issues.

1. Python lib 3,7 not included in the script. (easy one)
2. line 8 in modbusregister.py change it to def __init__(self, function, address,dataType,devide,sensorType, unit,UnitIdForIp,method,device,multiplydevice, digits=8):  Problem here was the wrong mix of var and string setting.
3. line 19 in modbusregister.py. change it to        if devide !="":
3 In the modbusdevices.ini the following Default is missing multiply=1
4 and the switch parameter methode isn't working correctly. Ficked it by setting is manual. It looks like it is capitals versus lowercase. 

Now the modbusdevice is starting up and the sensors are being made in domoticz. Now I still getting no data. 

I know the data is coming in (check it with a other script) . The module is techical working. 

Keep you posted on the work.

Regards, Koos.

[plugin - Copy.py.txt](https://github.com/DomoticX/domoticz-modbus/files/4575299/plugin.-.Copy.py.txt)
[modbusregister - Copy.py.txt](https://github.com/DomoticX/domoticz-modbus/files/4575300/modbusregister.-.Copy.py.txt)
[modbusdevices - Copy.ini.txt](https://github.com/DomoticX/domoticz-modbus/files/4575301/modbusdevices.-.Copy.ini.txt)

KoosBoersma, 2020-05-04:
For testing. This is the other script that I use. 
This updates values via JSON push to dummy created sensors. 

Easy in setup and is working.

Regards, Koos.
[miniPymodbus.old.py.txt](https://github.com/DomoticX/domoticz-modbus/files/4575343/miniPymodbus.old.py.txt)

DomoticX, 2020-05-05:
Fixed in update Modbus READ 2020.2A, if not please make another topic. closing this one...

closed 2020-05-05 by DomoticX