DomoticX/domoticz-modbus #25: Need to write Variables from Domoticz to Modbus Register

spnk0, 2020-05-19:
It would be very nice to set the Payload On/Off for Modbus TCP from Devices or Variables from within Domoticz, so one can publish sensor readings and all other values to ModbusTCP.

Regards,

Johannes

alexkiddde, 2021-01-27:
I would appreciate this too a lot. Just sending to different payload states does not fit many requirements.

Vincent835, 2021-02-06:
Hi,

There may be what you were looking for here:
https://www.domoticz.com/forum/viewtopic.php?p=256121#p256121

> Re: Python plugin: Modbus RTU / ASCII / TCP/IP
Post by eSbek » Saturday 12 September 2020 12:13

> Hey developers of the plugin,
> Some time ago I modded v1.0.9 for my house ventilation system to use different values of the fan speeds whenever I need (not using fixed payload value). So I decided to read a dedicated domoticz user variable inside the python script and write the value by Modbus to my ventilation system (eg. hi humidity detected in the bathroom -> speed goes on higher values).
> Please find the mod if you want to develop same approach in your latest plugins:
> https://pastebin.com/5zwiepyu