DomoticX/domoticz-modbus #28: Modbus-write RTU payload - register number

m-ur, 2020-09-16:
I'm using domoticz-modbus to communicate with PLC. When I was trying to send data by modbus write in RTU mode I always received data in the same registry. Changing register number field value haven't make any effect. I checked source and found this section:

`if (self.Domoticz_Setting_Modbus_Function == "5"): result = client.write_coil(int(self.Domoticz_Setting_Device_ID), int(payload, 16), unit=int(self.Domoticz_Setting_Device_ID))
if (self.Domoticz_Setting_Modbus_Function == "6"): result = client.write_register(int(self.Domoticz_Setting_Device_ID), int(payload, 16), unit=int(self.Domoticz_Setting_Device_ID))
if (self.Domoticz_Setting_Modbus_Function == "15"): result = client.write_coils(int(self.Domoticz_Setting_Device_ID), int(payload, 16), unit=int(self.Domoticz_Setting_Device_ID))
if (self.Domoticz_Setting_Modbus_Function == "16"): result = client.write_registers(int(self.Domoticz_Setting_Device_ID), int(payload, 16), unit=int(self.Domoticz_Setting_Device_ID))`

After changing first
`int(self.Domoticz_Setting_Device_ID)`
into 
`int(self.Domoticz_Setting_Register_Number)`
in each line I could write into registers at different addresses. I've tested only write function 6 - write into single holding register so other modbus writing functions still need testing.

Vincent835, 2020-09-16:
Hello,
Thank you.
You saved me.
I didn't understand why it didn't work.
With your correction it's perfect.

biwizit, 2021-01-09:
Thanks a lot @Vincent835 ! t is exactly the same issue for the TCP/IP part (from line 258 to 261):

```
if (self.Domoticz_Setting_Modbus_Function == "5"): result = client.write_single_coil(int(self.Domoticz_Setting_Device_ID), int(payload))
if (self.Domoticz_Setting_Modbus_Function == "6"): result = client.write_single_register(int(self.Domoticz_Setting_Device_ID), int(payload))
if (self.Domoticz_Setting_Modbus_Function == "15"): result = client.write_multiple_coils(int(self.Domoticz_Setting_Device_ID), [payload])     # TODO split up multiple bytes to proper array.
if (self.Domoticz_Setting_Modbus_Function == "16"): result = client.write_multiple_registers(int(self.Domoticz_Setting_Device_ID), [payload]) # TODO split up multiple bytes to proper array.
```

We also need to replace
`int(self.Domoticz_Setting_Device_ID)`
by
`int(self.Domoticz_Setting_Register_Number)`

It worked fine for me after the modification.

Vincent835, 2021-01-09:
Hi biwizit,

It was @m-ur who found the first  error.

biwizit, 2021-01-10:
Yes, you're right, thank you also @m-ur !

alexkiddde, 2021-01-28:
Hi @Vincent835,

commit eff45a5 does not fix the bug in all sections. As @biwizit mentioned correctly, the issue appears in the RTU as well as the TCP/IP part. I think this prevents many user from writing modbus values via TCP. 

Anyway, thanks @Vincent835 for this wonderful plugin and @m-ur for finding this issue. If I could have one wish, I would appreciate an option to write different values/variables in the modbus write plugin - not only two for on and off.

Vincent835, 2021-02-06:
Hi,

I modified ModbusWrite, in my fork, as suggested by biwizit. But I can't test it. If someone wants to do it and it works, I will suggest to Domoticx to integrate it into its repository.

alexkiddde, 2021-02-07:
> Hi,
> 
> I modified ModbusWrite, in my fork, as suggested by @biwizit. But I can't test it. If someone wants to do it and it works, I will suggest to @DomoticX to integrate it into its repository.

Works fine for me, thanks!

DomoticX, 2021-02-07:
Thanks all!, fixed in WRITE v2021.2

closed 2021-02-07 by DomoticX