DomoticX/domoticz-modbus #31: Error line 19 "devide"

jfh75, 2021-02-15:
Hello,

I am running domoticz on a ubuntu 20.04 standalone PC : 
Version: 2020.2 (build 12900)
Build Hash: 9cca890b4-modified
Compile Date: 2021-02-01 11:29:34
dzVents Version: 3.1.4
Python Version: 3.8.5 (default, Jul 28 2020, 12:59:40) [GCC 9.3.0]

I don't manage to make it work i have following error when trying to install SDM220 : 
 2021-02-15 08:18:18.016 Error: Python Plugin System: (Modbus2) failed to load 'plugin.py', Python Path used was '/home/jfh/domoticz/plugins/modbus-device/:/usr/lib/python38.zip:/usr/lib/python3.8:/usr/lib/python3.8/lib-dynload:/usr/local/lib/python3.8/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.8/dist-packages'.
2021-02-15 08:18:18.016 Error: Python Plugin System: (eastron) Module Import failed, exception: 'SyntaxError'
2021-02-15 08:18:18.016 Error: Python Plugin System: (eastron) Import detail: File: /home/jfh/domoticz/plugins/modbus-device/modbusregister.py, Line: 19, offset: 19
2021-02-15 08:18:18.016 Error: Python Plugin System: (eastron) Error Line ' if devide<>"":
2021-02-15 08:18:18.016 '
2021-02-15 08:19:27.883 Error: eastron hardware (24) thread seems to have ended unexpectedly
2021-02-15 08:19:41.886 Error: eastron hardware (24) thread seems to have ended unexpectedly
2021-02-15 08:19:55.888 Error: eastron hardware (24) thread seems to have ended unexpectedly 

Is there something wrong with my install ?

Thank you for your kind help

JFH

xbeaudouin, 2021-05-03:
Hi, 
Seem the plugin use some old python code.
I have tried to fix it, but still need work -> see https://github.com/xbeaudouin/domoticz-modbus/tree/master/modbus-device
Needs some more help to make this working as well...

AdamWeglarz, 2023-03-26:
any news on that. I would like to use this plugin to read values from LUMEL PVSA.