DomoticX/domoticz-modbus #37: Modbus error after update version 2022.2

T1NBK, 2022-11-12:
After updating Domoticz 2022.1 to 2022.2 I get a lot of the following errors

"Modbus error decoding or received no data (TCP/IP)!, check your settings!"

More people with this issue?

T1NBK, 2022-11-13:
I think I have the cause of the problem. I have quite a bit of modbus sensors in Domoticz. When I reboot the system Domoticz initiates all these sensors at the same time. Creating a storm of requests to my PLC's at the same moment.
When I disable the sensors in the hardware panel of Domoticz and then enable them again one by one they are no longer in sync and the problem disapears.

kawiarz33, 2022-11-17:
I have also problem with this plugin but I communicate by Modbus RTU - READ with heat pump. After upgrade of domoticz to 2022.2 I can't get data from the controller, when I restart the system it can read 2 or 3 variables out of 12 but it works till next read and then again it is blocked.
Power metering plugin for chint dtsu666 based on minimalmodbus works properly

Domoberry, 2023-03-15:
This plugin has been working fine up to the moment I upgraded to Domoticz 2023.1 (from 2022.1, also several other updates, all on Buster): I now also see multiple ""Modbus error decoding or received no data (TCP/IP)!, check your settings!" errors. I'm reading 8 values from a energy meter at 30 seconds polling. Reduced to 5: some errors remaining. Re-enabled each of the 5 one minute apart as suggested T1NBK, some errors remaining. I tried using slightly different polling intervals for each of the 5: still these errors. The plugin.py code has 5 locations where this error can be triggered. Tagged them to see that the sections 'Decode Data Type' and 'Decode Data Value' generated them. And that is where my Python knowledge stops...
Calling on experts! (since I'm not aware of any other Modbus plugin for Domoticz).

sebekhtc, 2023-12-12:
I have always had these errors but could read the values outside errors (the Domoticz variable is just not updated when error occur and last good value is there). The device I poll has a limit of polls per seconds so I believe that causes the error, teh plugin receives no value and can't decode it. I tried to change timings but can't see improvements, same time I can't (don't know how to) verify the timings are correct, maybe there is the issue? Anyway I have been running the plugin for years with these errors and my device I control works as expected despite the errors inbetween polls.

Domoberry, 2023-12-12:
@sebekhtc thanks for the reply. In my case some of the modbus measured values also arrived OK in Domoticz. Yet, there were mutliple errors logged as mentioned above. The error message can have btw several causes.
Based on reading some other postings, here is my understanding (which might be incorrect/incomplete :-) ):
The root cause is a change in the way Domoticz handles plugins. In my case, each instance of the (very nice and complete) Domotix plugin takes care of reading one Modbus value. I had multiple instances of this plugin as I wanted to obtain multiple values via Modbus. After the Domoticz change, these multiple instances of the Domotix plugin were competing for access to the port for the Modbus hardware interface at the same time. This causes the problem. Restarting the Domotix plugins one-by-one after each other improves the situation as the change of a clash reduces, Using only one Domotix plugin works fine obviously.
For me the solution was to take a different (less sophisticated) plugin https://github.com/MFxMF/SDM630-Modbus as starting point and modify the plugin to work with my Modbus Energy meter (a Sinotimer DDS024MR 3 phase) picking up the measurement values of interest. Basically, a single instance of this plugin obtains all values of interest in a single communication session with the Modbus device. So, I only need one instance of the plugin in my case. Clearly, this does not address the root cause, yet having a single instance of the Modbus plugin getting all values is an acceptable workaround for me. Working OK for a few months now.