DomoticX/domoticz-modbus #38: modbus.client.sync not found

Micha-123, 2022-11-13:
packages are installed with

sudo pip3 install -U pymodbus pymodbusTCP

but still it not works and means "not found" 

using last raspbian

2022-11-13 10:42:20.650 Error: Epever Batt Volt: (ModbusREAD) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/modbus-read/:/usr/lib/python39.zip:/usr/lib/python3.9:/usr/lib/python3.9/lib-dynload:/usr/local/lib/python3.9/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.9/dist-packages'.
2022-11-13 10:42:20.659 Error: Epever Batt Volt: Traceback (most recent call last):
2022-11-13 10:42:20.660 Error: Epever Batt Volt: File "/home/pi/domoticz/plugins/modbus-read/plugin.py", line 177, in <module>
2022-11-13 10:42:20.660 Error: Epever Batt Volt: from pymodbus.client.sync import ModbusSerialClient # RTU
2022-11-13 10:42:20.660 Error: Epever Batt Volt: ModuleNotFoundError: No module named 'pymodbus.client.sync'
2022-11-13 10:54:47.470 Error: Epever Batt Volt: (ModbusREAD) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/modbus-read/:/usr/lib/python39.zip:/usr/lib/python3.9:/usr/lib/python3.9/lib-dynload:/usr/local/lib/python3.9/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.9/dist-packages'.
2022-11-13 10:54:47.480 Error: Epever Batt Volt: Traceback (most recent call last):
2022-11-13 10:54:47.480 Error: Epever Batt Volt: File "/home/pi/domoticz/plugins/modbus-read/plugin.py", line 177, in <module>
2022-11-13 10:54:47.480 Error: Epever Batt Volt: from pymodbus.client.sync import ModbusSerialClient # RTU
2022-11-13 10:54:47.480 Error: Epever Batt Volt: ModuleNotFoundError: No module named 'pymodbus.client.sync'

x-rav66, 2022-11-15:
try:
sudo pip3 uninstall -U pymodbus
sudo pip3 install -U pymodbus==2.5.3

simat-git, 2023-04-30:
The latest pymodbus has changed , if think since Python 3.7, so just change

from pymodbus.client.sync import ModbusSerialClient # RTU
from pymodbus.client.sync import ModbusTcpClient    # RTU over TCP

to

from pymodbus.client import ModbusSerialClient # RTU
from pymodbus.client import ModbusTcpClient    # RTU over TCP

pabaillon, 2024-02-12:
> The latest pymodbus has changed , if think since Python 3.7, so just change
>
> from pymodbus.client.sync import ModbusSerialClient # RTU from pymodbus.client.sync import ModbusTcpClient # RTU over [...]

This doesn't work but "sudo pip3 install -U pymodbus==2.5.3" works. THanks

diliak, 2024-11-05:
Yes, working for me on latest Domoticz 2024.7 and python 3.9 on Debian 11.

With : 

`
sudo pip3 uninstall pymodbus
sudo pip3 install -U pymodbus==2.5.3
`