DomoticX/domoticz-modbus #40: failed to load 'plugin.py'

bonebuster, 2023-12-13:
Hi
i have this issue (ModbusREAD) failed to load 'plugin.py', Python Path used was '/home/pi4/domoticz/plugins/modbus-read/:/usr/lib/python39.zip:/usr/lib/python3.9:/usr/lib/python3.9/lib-dynload:/usr/local/lib/python3.9/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.9/dist-packages'.

Domoticz 2023.2 / Bullseye

have you an idea  to help me ?

bonebuster, 2023-12-18:
there is no one to helpme ?

bonebuster, 2024-01-17:
Am I the only one with this problem?

DomoticX, 2024-01-17:
Have you looked at https://github.com/DomoticX/domoticz-modbus/issues/38 ?, maybe same issue and how to resolve it.

bonebuster, 2024-01-17:
Hi 
I tried but it doesn't work
I have the same errors
it seems that some things have changed and the files are not in the right place

Anyone who try with new install maybe have same issue

DomoticX, 2024-01-17:
And you did install pymodbus and pymodbusTCP?

Hmmm is there any other log available (to see the underlying problem)?, what if you run the plugin.py with python3 from bash?, any see if any errors are popping up (e.g. import errors)?

bonebuster, 2024-01-18:
Hi

Yes pymodbus and pymodbusTCP are installed with pip install

I have done a lot of installations since December
bullseyes, domoticz, pip, pymodbus, pymodbusTCP

on pi3b, pi4b,VMs
always make the same error

I'll try to run plugin.py this day with python3 to see other log
Many thanks

bonebuster, 2024-01-18:
Hi
here's my log

[image]

DomoticX, 2024-01-21:
Hmmm nothing much on info in that error, when i find some time later-on i will give the new domoticz a try!, Very strange no one else is here to help....

sebekhtc, 2024-08-04:
Im trying to configure Chromebox 3 and move Domoticz there, so far I installed the latest pymodbus and got similar error. it was gone when I changed
from pymodbus.client.sync import ModbusSerialClient # RTU
from pymodbus.client.sync import ModbusTcpClient    # RTU over TCP

to 
from pymodbus.client import ModbusSerialClient # RTU
from pymodbus.client import ModbusTcpClient    # RTU over TCP

as the  pymodbus.client.sync no longer exist it is now pymodbus.client.

Next there was an error with these lines:
            client = ModbusTcpClient(host=self.Domoticz_Setting_TCP_IP, port=int(self.Domoticz_Setting_TCP_PORT), framer=ModbusRtuFramer, auto_open=True, auto_close=True, timeout=2)
 
seems like auto_open and auto_close is not used anymore at RTU over TCP and TCP/IP as in the library it looks:
```
self.comm_params = CommParams(
                comm_type=CommType.TCP,
                host=host,
                port=port,
                comm_name=name,
                source_address=source_address,
                reconnect_delay=reconnect_delay,
                reconnect_delay_max=reconnect_delay_max,
                timeout_connect=timeout
```

the plugin looks now (has to be changed in both read and write):
  ```
      ########################################
        # SET HARDWARE - pymodbus: RTU over TCP
        ########################################
        if (self.Domoticz_Setting_Communication_Mode == "rtutcp"):
          Domoticz.Debug("MODBUS DEBUG - INTERFACE: IP="+self.Domoticz_Setting_TCP_IP+", Port="+self.Domoticz_Setting_TCP_PORT)
          Domoticz.Debug("MODBUS DEBUG - SETTINGS: Method="+self.Domoticz_Setting_Communication_Mode+", Device ID="+self.Domoticz_Setting_Device_ID+", Register="+self.Domoticz_Setting_Register_Number+", Function="+self.Domoticz_Setting_Modbus_Function+", Data type="+self.Domoticz_Setting_Data_Type+", Pollrate="+self.Domoticz_Setting_Device_Pollrate)
          try:
#            client = ModbusTcpClient(host=self.Domoticz_Setting_TCP_IP, port=int(self.Domoticz_Setting_TCP_PORT), framer=ModbusRtuFramer, auto_open=True, auto_close=True, timeout=2)
            client = ModbusTcpClient(host=self.Domoticz_Setting_TCP_IP, port=int(self.Domoticz_Setting_TCP_PORT), framer=ModbusRtuFramer, timeout=2)
          except:
            Domoticz.Error("Error opening RTU over TCP interface on address: "+self.Domoticz_Setting_TCP_IPPORT)
            #Devices[1].Update(1, "0") # Set value to 0 (error)

        ########################################
        # SET HARDWARE - pymodbusTCP: TCP/IP
        ########################################
        if (self.Domoticz_Setting_Communication_Mode == "tcpip"):
          Domoticz.Debug("MODBUS DEBUG - INTERFACE: IP="+self.Domoticz_Setting_TCP_IP+", Port="+self.Domoticz_Setting_TCP_PORT)
          Domoticz.Debug("MODBUS DEBUG - SETTINGS: Method="+self.Domoticz_Setting_Communication_Mode+", Device ID="+self.Domoticz_Setting_Device_ID+", Register="+self.Domoticz_Setting_Register_Number+", Function="+self.Domoticz_Setting_Modbus_Function+", Data type="+self.Domoticz_Setting_Data_Type+", Pollrate="+self.Domoticz_Setting_Device_Pollrate)
          try:
#            client = ModbusClient(host=self.Domoticz_Setting_TCP_IP, port=int(self.Domoticz_Setting_TCP_PORT), unit_id=int(self.Domoticz_Setting_Device_ID), auto_open=True, auto_close=True, timeout=2)
            client = ModbusClient(host=self.Domoticz_Setting_TCP_IP, port=int(self.Domoticz_Setting_TCP_PORT), unit_id=int(self.Domoticz_Setting_Device_ID), timeout=2)
          except:
            Domoticz.Error("Error opening TCP/IP interface on address: "+self.Domoticz_Setting_TCP_IPPORT)
            #Devices[1].Update(1, "0") # Set value to 0 (error)
```
I'm dealing now with:
Modbus error communicating! (RTU/ASCII/RTU over TCP), check your settings!

diliak, 2024-11-05:
Hello,
Trying to use your modified files on Domoticz 2024.7 with usb to 485 plug.
Created a Modbus RTU / ASCII / TCP/IP - READ v2021.7 on hardware

Error :

 2024-11-05 10:28:04.079 Error: test: (ModbusREAD) failed to load 'plugin.py', Python Path used was '/home/diliak/domoticz/plugins/modbus-read/:/usr/lib/python39.zip:/usr/lib/python3.9:/usr/lib/python3.9/lib-dynload:/usr/local/lib/python3.9/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.9/dist-packages'.
2024-11-05 10:28:04.088 Error: test: Exception: 'SyntaxError'. No traceback available. 

And every 10 sec approx :

2024-11-05 10:31:03.620 Error: test hardware (9) thread seems to have ended unexpectedly

Any ideas ?